mcqn/my-bikes-got-led #6: Find process for connecting wires to the light strips

amcewen, 2021-12-20:
The WS2812B lights come in reels of 5m.  A bike will need ~1/3rd of that, so there'll be two strips which will need headers.

The system used for the Geometric Lights would work, but is quite labour-intensive.  It'd be nice to find a simpler option.

amcewen, 2022-05-16:
We've  got some supposedly clip-in connectors, so we should work out how well they work.

huffeec, 2022-10-21:
The clips seem to be 10mm and the strips 12mm. 
I've tried lots of methods to connect these. The best approach so far is to trim away the sides of the sheath so that the strip fits into the clip and putting a small piece of double sided foam tape on the top of the strip to provide enough thickness that the pins in the clip and puncture the bottom plastic. I also had more success taking the clip completely apart so that it's easier to seat the strip and then adding the top of the clip.

amcewen, 2022-10-31:
For the batch we prepped for the Katumba Halloween Carnival we soldered wires to the strips and sealed them with 12.7mm ID glue-lined heatshrink.  Then the other end could be cut to length and a cap glued on with rubber cement.