mcqn/my-bikes-got-led #8: Connect 2 additional gpio pins to level shifter

huffeec, 2022-10-10:
Include breakable jumpers so it can be turned off

huffeec, 2022-10-12:
V2 board includes access to pins IO5 and IO16

JackiePease, 2022-10-27:
This is the [link to multi-strip support info](https://kno.wled.ge/features/multi-strip/).
It says that "It is highly recommended to use two specific LED pins, GPIO1 (TX) and GPIO2 (D4), since they allow for hardware driving." although others can be used with the bitbang method.

huffeec, 2022-10-31:
That's useful to see. Given that, "It is highly recommended to use an ESP32 when using more than 1 output." is this something we need to be too concerned with while we're using 8266's?

huffeec, 2022-11-01:
Checking this IO2 is already available as the led_data is there value in breaking out more for this rev?

amcewen, 2025-01-03:
This has been folded into the v2 board updates.

closed 2025-01-03 by amcewen