mcqn/my-bikes-got-led #9: Stock for new boards

huffeec, 2022-10-19:
There are 15 unpopulated v1 pcbs 
There are enough components in stock to populate 9 boards

To populate fifteen boards the following additional components are needed
C3, C4, C6, C8 - 22uF (1812) Short by 12
R5, R14 - 1K (0603) Short by 1
R1, R2, R10 - 10K (0603) short by ~30
R13 -7K5 (0603) short by 1
R15 - 90K (0603) short by 1
R16 - 20K (0603) short by 1
U4 - MT3608 SOT-23-6 short by 1

I've not managed to find any of the 22uH inductors (L1) yet or the battery protection IC(U3) but suspect there are as many remaining as the other inductor (15)? 

L1 - 22uH 
U3 - FS312F-G

huffeec, 2022-10-19:
[Sheet with full stock list of components](https://github.com/mcqn/my-bikes-got-led/blob/main/electronics/MBiGL_Stock.ods)

amcewen, 2022-10-19:
Parts ordered to make up the shortfall, apart from U4 as Farnell don't stock it.

amcewen, 2022-10-21:
Parts arrived yesterday.  I think we'll leave U4 until we're ordering more parts from China.

closed 2022-10-21 by amcewen