mcqn/my-bikes-got-led #12: v2 boards updates

huffeec, 2022-11-09:
Changes for V2

Update the light connector to use [https://uk.farnell.com/amphenol-anytek/oq0412510000g/terminal-block-r-a-header-4way/dp/3810182](https://uk.farnell.com/amphenol-anytek/oq0412510000g/terminal-block-r-a-header-4way/dp/3810182)

Added charging circuit from [Tactile Electronics](https://github.com/Tactile-electronics/windmill-electronics)

Added connection to fix #2 added wire is between FS312F-G (U3) pin 2 and JSM2302 (Q2) pin 3

reduced board height from 90mm to 82mm

amcewen, 2022-11-17:
Reviewed the changes:
* We should use thicker traces for the power circuits
* The `U3` and `Q1` (but not `Q2`) protection circuit should probably move to just by the `BT1` and `J9` area, so it's protecting the battery, rather than after the charging side
* Can the regulators - both 5V and 3V3 - cope with 5V as input, given that the charging circuit will supply it.  Might need to provide some protection for the solar input too, as that could provide even higher voltages :worried:   Actually, reading the datasheet for the 5V boost chip, its maximum input voltage is 4.5V, so we might need protection for just USB, never mind higher solar voltages.  Not sure how useful the running-while-charging approach will be, given the current the LEDs can draw will often be beyond what the USB will provide, but it'll be good to try (and I suppose if nothing else, maybe it'll just power the lights so less is drawn from the battery).  A [zener diode regulator circuit](https://wiraelectrical.com/zener-diode-voltage-regulator/) might be enough?
* Checking USB-C connector availability, I see [Farnell has them](https://uk.farnell.com/amphenol-fci/12401610e4-2a/usb-conn-3-1-type-c-rcpt-smd/dp/3367461) which is good (might be able to find cheaper, but I wanted to check they were readily available)
* I wonder if `L2` and `L3` need to be so *physically* big?  Smaller ones would free up a bit of board space, but it's probably not worth changing unless we end up revisiting the area where they are and some more room would be handy...
* There are no mounting holes on the PCB at present.  I think that's because they're easy to end up deleting when you import a new schematic as they're footprints that don't have an associated symbol.  Similarly with the assorted logos: WEEE, RoHS, OSHWA, etc.  There are "Graphic" symbols it seems, which we should add to the schematic so they'll pull in the right logos and make sure they don't get lost.  Seems there are "Mechanical" ones too, so we can use those to add the mounting holes.

huffeec, 2022-12-12:
Above updates addressed in newest commit.
The ERC and DRC have both been run, akaik the remaining errors are not significant.

amcewen, 2022-12-20:
* We should check the battery usermod in case we need to tweak the battery->ADC circuitry
* Revision number needs updating
* Can we squeeze on a smaller MCQN logo in the corner by the FTDI connector?
* Would be good to label the jumpers, but not sure if there's space
* Do we need any thermal vias on the boost or charging circuit?

huffeec, 2022-12-21:
The latest update has points 2, 4 and 5** complete.
** I added vias and extra copper around the components. There wasn't much of a clue about layout in the datasheet so this is all fairly arbitrary and could use a sense check.

1 Short answer to this is I'm not sure. It feel like this must need to be a potential divider circuit to bring down the voltage to ~3 max. Which would suggest we need a 2nd resistor in series with R20. Related to this R28 value is still tbc and may make the second resistor redundant in the battery monitor circuit.

3 Adding logos needs to be easier in kicad but I'm sure I can squeeze something in here.

amcewen, 2023-01-18:
Latest revision looks on the brink of good to go.  Can we add the MCQN, WEEE, and RoHS logos onto the underside of the PCB.  Then if you can export the Gerbers we can kick off a production run :grin:

huffeec, 2023-01-18:
Output file added with Gerbers

huffeec, 2023-02-08:
First boards populated.
Bugs:

Track from Q3 pin 3 to Q2 pin 1 should have a junction where it crosses the -Batt line
(Temp fix is a wire from jp6 to -Batt (bt1)
Remove R28? 
Some board hum and warming of L2 with 100 lights running, needs further investigation

huffeec, 2023-02-13:
The fix above q3 p3 to q2 p1 will work with battery power but not by usb. The switch circuit still isn't working.
While this appears to work I think this is only related to the solution

Have tried removing this wire and instead connecting q2 p3 to u6 p6  
D4 lights with usb but no other activity
Nothing if powered by battery

It appears the protection IC is still where the error is. Currently Q2 is not operating, OD low/OC high and the CS (p2) path is incorrect. (This should probably route to GNG (p6)

Next steps
[Confirm CS path from datasheet](https://www.ic-fortune.com/upload/Download/FS312F-G-DS-12_EN.pdf)

Split the sheets into modular components 
Battery + protection
Charging
Boost conversion
Level shifting
ESP

huffeec, 2023-02-15:
Potential issues seen when creating a more modular schematic.

The Battery Mod addition uses a PD between Batt+ and Gnd to provide the A0 input. This could explain why the switch is being bypassed?
The 1k resistor on the FS312 between CS and Gnd is missing

amcewen, 2023-02-22:
We should check whether the charging circuit ever gets to "done", given the Tactile Electronics board we've cribbed it from [always shows the "charging" LED](https://github.com/Tactile-electronics/windmill-electronics/issues/13)

amcewen, 2023-06-07:
Hopefully we'll be able to use an updated version of the Tactile Electronics board, but if we haven't solved it by July then we should look at switching to just a USB-charging option for the v2 boards (and we can add solar and/or dynamo charging in the future)

amcewen, 2023-11-23:
I'm wondering, although it'll be 50p or so more expensive on the BoM, if we should switch to [this TI combined charging and boost converter chip](https://www.ti.com/lit/ds/symlink/bq25895m.pdf) particularly because that can do faster charging from USB and will provide up to 3A of 5V

amcewen, 2025-01-03:
This has been superseded by #20

closed 2025-01-03 by amcewen