mcqn/my-bikes-got-led #13: Board reworking

huffeec, 2022-11-09:
Of the remaining 8 V1 boards;

1 is working and running WLED
2 need the firmware flashing **
5 need further investigation

amcewen, 2022-11-19:
Had a look at some more of the boards today.
* One with no power to the ESP had the 3V3 fuse only fully soldered at one side, and the inductor was similarly only soldered down on one side.  Now it seems to be functioning okay - I've put code on it and checked the AP is responding, but didn't have LEDs to hand to test it fully
* One with no comms to the ESP had the reset button not soldered down properly.  Soldering that up seems to have fixed things - not completely sure why, as it shouldn't have affected anything.  Beforehand I was receiving data from it over serial, but didn't seem to be able to send any data to it.  Now has code uploaded fine and the AP checked.  Needs testing fully with LEDs.
* The one with code but no AP doesn't seem to be booting up correctly.  I only ever get the same boot message, whereas I should be able to put it in programming mode by grounding pin `0` (which would give a different boot mode message).  I tried resoldering the resistors which should set some of that, but it hasn't made any difference.  Haven't spent lots of time on it, so we should check the docs and the connections on the ESP accordingly.

amcewen, 2022-11-25:
Checked the `boot mode` messages for the third board listed above.  It was consistently coming up with `boot mode:(1,0)`, which *is* programming mode.  For a "normal" boot the first digit would be `3`.  Pin `0` was permanently shorted to `GND`.  Some flux and heat on the `0` and `GND` pair cleared that up so it's working now.

huffeec, 2022-12-14:
Two more boards added to the OK pile today with new  U5 step down convertors (3v3)

amcewen, 2023-02-22:
Have we rescued all these boards now?  (Or at least all the ones worth trying to rescue?)  Can we close this off?