mcqn/my-bikes-got-led #14: Add voltage divider on battery voltage monitor

amcewen, 2022-11-25:
Checking over a board now I realised that the battery voltage is just wired to the analogue input of the ESP8266, so it's usually above 3.3V.  We haven't ever used it - it was more of a "this might be useful" addition, and there's a jumper to cut it, but we should check the voltages that the input can cope with and look at a voltage divider if it's out of range when the battery is 4.2V.

amcewen, 2022-12-07:
@huffeec pointed out there's already one resistor in the circuit for measuring battery voltage, which reminded me that I'd done some initial investigation of it when designing the v1 board.  The circuit we have now is the one that the [battery_status_basic](https://github.com/Aircoookie/WLED/tree/main/usermods/battery_status_basic) usermod suggests, so we should try enabling that and seeing what happens :grin:

huffeec, 2023-01-04:
Enabling the mod on a v1 board gives a reading of full battery, regardless of the actual status.This is a new link for the usermod info, it's changed above [https://github.com/Aircoookie/WLED/tree/main/usermods/Battery#useful-links](https://github.com/Aircoookie/WLED/tree/main/usermods/Battery#useful-links)

The [circuit described](https://arduinodiy.wordpress.com/2016/12/25/monitoring-lipo-battery-voltage-with-wemos-d1-minibattery-shield-and-thingspeak/) is based on a [Wemos D1](https://www.wemos.cc/en/latest/_static/files/sch_d1_mini_v3.0.0.pdf) which has an 'internal' divider circuit.

This isn't something currently on the mbigl board or internal to the ESP (afaik). [ESP Wroom 02](https://www.espressif.com/sites/default/files/documentation/0c-esp-wroom-02_datasheet_en.pdf)
The reversed zener diode (D8) is keeping the +Batt voltage in the range 0 - 4.3V  

We need to be in the 0-1V range for the ADC so my (current) thinking is that that we will need a full potential divider around jp4 for the battery usermod to work.

I think...

amcewen, 2023-01-09:
Makes sense.  Reading through the circuit description, we'd want to have a voltage divider that's BatteryVoltage -> 320K resistor -> A0 -> 100K resistor -> Ground.

huffeec, 2023-01-11:
Added

closed 2023-01-11 by huffeec