mcqn/my-bikes-got-led #19: Upgrade to an ESP32

amcewen, 2024-07-14:
The ESP8266 is starting to lag too far behind the WLED capabilities.  Although we've still got lots in stock, maybe we can leave them for the Ackers Bell or Mersey Dot?

Anyway, we should move to one of the cheap ESP32 modules, and switch to the "mini" version to make the board smaller.

amcewen, 2024-07-14:
I'd been assuming we'd go with an ESP32-C3 **but** thinking about the new Bluetooth Auracast tech (the broadcast audio option), having a module that supports *that* would open up some interesting possibilities with the sound-reactive versions.

The Auracast stuff needs Bluetooth 5.2 or higher, which means we should go with the ESP32-C6-MINI-1-N4 over the C3

amcewen, 2024-11-06:
Tested the latest WLED on a C3 dev board, hooked up with a I2S microphone (an INMP441 on a breakout board this time) as [detailed in the audio-reactive guide on the WLED wiki](https://kno.wled.ge/advanced/audio-reactive/) and it seems perfectly responsive with ~150 LEDs.

So, given [support for the C6 will be some time away](https://github.com/Aircoookie/WLED/issues/3078) (and we might be able to upgrade the Bluetooth stack to support auracast on older boards, maybe?) we're going with a C3-MINI for this rework.

huffeec, 2024-11-06:
Looking at the schematic for the C3 there are less GPIOs than we need

Of the [15 available](https://www.espressif.com/sites/default/files/documentation/esp32-c3-mini-1_datasheet_en.pdf)

- 2 (IO18/19) are D+/-
- 2 rx/dx
- 3 IO2/08/09 are strapping pins for boot mode

Section 4.1 Chip Boot Mode Control says that "GPIO2 actually does not determine SPI Boot and Joint Download Boot mode, but it is recommended to pull this pin up due to glitches"

IO8 (d/c SPI Download High)

IO9 needs to be accessible (High SPI Low Download) 

Leaving 8 accessible pins, the current design uses: 

- 4 for level shifted data channels
- 2 for leds
- 2 pads for optional external crystal
- 1 for reset button
- -1 remaining

To provide I2S we need 3-4 pins available

We could use the two crystal pins but are still 3 pins short that we need to find from somewhere. Either from strapping pins (It seem odd to me that these are never available, or using less leds/data channels)

IO0 and IO1 are currently exposed as pads in case an external crystal is needed

huffeec, 2024-11-06:
[This page from Espressif](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/boot-mode-selection.html) seems to suggest that 08 needs to be pulled high in hardware which matches the [C3 mini data sheet example circuit - section 9
](https://www.espressif.com/sites/default/files/documentation/esp32-c3-mini-1_datasheet_en.pdf)

The page linked [refers to a dev kit schematic](https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch-20180607a.pdf) that doesn't appear to have an IO8

huffeec, 2024-11-06:
For reference and to aid my memory, the [C3 mini dev kit schematic](https://dl.espressif.com/dl/schematics/SCH_ESP32-C3-DEVKITM-1_V1_20200915A.pdf) has 08 pulled high but not 02. Although the schematic is dated 2021 and the datasheet was updated in 2024 (V1.7) to change the boot pin requirements in 2024 so I suspect both are required

huffeec, 2024-11-06:
Tasks remaining
- [ ] Update references in schematic
- [ ] Reduce edge cut to width of <48mm in pcb
- [ ] Check for smaller footprint components eg L2/passives
- [ ] Add second unpopulated usb socket
- [ ] Retrace tracks
- [ ] Update silkscreen
- [ ] run DRC
- [ ] run DRC
- [ ] run DRC

amcewen, 2025-01-03:
The ESP32C3 on the prototypes seems to be working well, so closing this off

closed 2025-01-03 by amcewen