mcqn/my-bikes-got-led #21: Pin access for redesign

huffeec, 2024-08-06:
With the new components #20 we will be able to slightly reduce the footprint of the board. With that in mind do we still want to expose all the pins of the esp32, provide a subset, or none at all?

The ESP32-C6-MINI-1-N4 #19 has USB serial, can we lose the UART header?

amcewen, 2024-08-07:
I think having a few pins available for other things is useful, but it doesn't need to be the full set.  Enough for a button (in WLED, so in addition to the usual bootstrapping ones), and an ADC and I2S pins to allow either option for the sound-reactive options (if the ADC can use one of the I2S pins that'd be fine, as you won't want both options at the same time)

The USB is an interesting question.  I'd not spotted it has that.  The tricky part, I think, is that the battery charger chip also wants the USB D+/D- pins.  I think that's to allow faster charging options, which would (probably) trump easier flashing?  Most of the time you could flash new versions over WiFi.

huffeec, 2024-08-07:
The DSEL pin can be used to switch use of D+/- pins as described [here](https://e2e.ti.com/support/power-management-group/power-management/f/power-management-forum/1020519/bq25895m-reference-circuit-for).  

[reference circuit](https://www.ti.com/lit/ug/sluuba2b/sluuba2b.pdf )

But, this might be overcomplicating and not something we need to worry about?

amcewen, 2024-08-08:
Using the DSEL pin is tempting.  Looking at the circuit it'd be two extra components: a 2-channel switch and a pull-up resistor?

The TI switch is only available from Farnell with a 40-week lead time, but maybe [this NXP chip](https://uk.farnell.com/nxp/nx3dv42gu-115/analogue-switch-2-ch-dpdt-xqfn/dp/2749612RL) would be a suitable option?  That'd add ~43p to the BoM cost, which doesn't sound terrible for keeping USB support on the ESP32.

huffeec, 2024-08-08:
There isn't anything specific in the [c6 mini datasheet ](https://www.espressif.com/sites/default/files/documentation/esp32-c6-mini-1_mini-1u_datasheet_en.pdf)
but the [c6 series datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-c6_datasheet_en.pdf) does show it's supports USB 2 but not 3

"USB 2.0 full speed compliant, capable of up to 12 Mbit/s transfer speed (Note that this controller does
not support the faster 480 Mbit/s high-speed transfer mode)"  4.2.1.6

huffeec, 2024-08-08:
The [NXP NX3DV42GU-115](https://www.lcsc.com/product-detail/Analog-Switches-Multiplexers_NXP-Semicon-NX3DV42GU-115_C2652483.html) is available at LCSC in single units ($0.82ea).

amcewen, 2024-08-08:
Sounds like USB multiplexing is a go :grinning:

huffeec, 2024-10-18:
This has moved on because of #25 (Using the IP5310 instead)

The current design has d+/- direct to the ESP as the battery management isn't using them.

Additional inputs available 

- GPIO 18 and 19 are available through the spare channels of the level shifter 
- 9 pin header
-- GPIO 02 is available as ADC input
-- 8, 9, 22, 23 also available as spare GPIO inputs
-- GPIO 06/07 available as SDA SCL 
-- Remaining pins are 3v3 and GND
- RX, TX, GND and 5V (solder jumper) on a 6 pin header
- 2 x 3V3, 5V, GND
- Batt +/- as 2 pin head
- 5V, Data. GND 3pin header

5V, Data and GND test points
Test points on reverse for GPIO0/1 (External crystal)

There is also a test point on the battery management for the KEY pin in case we need that. And 3 unpopulated resistors VSET, RSET, VTHS. I don't think we'll need access to any other pins but there is space on the board if we do want to add extra test points for access

huffeec, 2024-10-21:
Closed with [945ffdf](https://github.com/mcqn/my-bikes-got-led/commit/945ffdfe1df8e9bcc706f4c1dc4ae111a14a96aa)