mcqn/my-bikes-got-led #24: Boosted output is not available if USB input is connected (Battery charging)

huffeec, 2024-08-21:
The PMID pin of the BQ25895M chip provides the boosted output providing the OTG pin is high and Vbus is disconnected.

So with an input at the usb port the device will provide power management to use this input and/or the battery to give the system voltage. 
However it can only provide the boosted output when the battery isn't charging (USB input removed) [8.2.5](https://www.ti.com/lit/ds/symlink/bq25895m.pdf)

This means we can operate lights only if a battery is connected but not a usb source eg power bank. 

Can we live with this or do we need to find a different solution?

.

amcewen, 2024-08-21:
If there's power on Vbus then we could route that to the lights, no?  That'll be more efficient than dropping it to battery voltage and then boosting it back up anyway.  That's what the last design did, with an extra mosfet to switch the boost output off when there was external power.  So we might need to replicate that, but it also sounds a bit like the internal `RBFET` might perform that role?

huffeec, 2024-10-18:
Fixed by #25