mcqn/my-bikes-got-led #25: Alternatives to BQ25895M

huffeec, 2024-09-25:
Alternatives to the TI battery management chips, there are many options that would reduce the BOM costs. It's not obvious to me what the right choice is. Here are a couple to start. 

[BQ25895M](https://www.lcsc.com/product-detail/Battery-Management_Texas-Instruments-BQ25895MRTWR_C485917.html)
[Boost output up to 3.1A @5V](https://www.lcsc.com/datasheet/lcsc_datasheet_2001181534_Texas-Instruments-BQ25895MRTWR_C485917.pdf)
LCSC costs
1+ $1.8218
10+ $1.6962

[INJOINIC IP5310](https://www.lcsc.com/product-detail/PMIC-Battery-Management_INJOINIC-IP5310_C191176.html/)
[Boost output up to 3.1A @ 5V](http://www.injoinic.com/wwwroot/uploads/files/20200221/a7a6128e03ee193cf35e48b28e3df9df.pdf)
LCSC costs
1+ 	$0.5921
10+  $0.4979

[ETA9740](https://www.lcsc.com/product-detail/Battery-Management-ICs_etasolution-ETA9740E8A_C7465512.html)
[Boost output up to 2.4A @5V](https://www.lcsc.com/datasheet/lcsc_datasheet_2307171444_etasolution-ETA9740E8A_C7465512.pdf)
5+ 	$0.2242
50+ $0.1782

amcewen, 2024-09-25:
From a sweep of my almost-lost-tabs...

[INJOINIC IP5322](https://www.lcsc.com/product-detail/Battery-Management_INJOINIC-IP5322_C188807.html)
[Boost output up to 3.1A @ 5V](http://www.injoinic.com/wwwroot/uploads/files/20200221/277b7d7f99700b3307be2cda1ca7000f.pdf)  Not quite sure why I'd saved that, though can (it seems) output 9V or 12V too
LCSC costs
1+ $0.8814
10+ $0.8629

[ETA9741](https://www.lcsc.com/product-detail/Battery-Management_etasolution-ETA9741E8A_C7465513.html)
[Boost output up to 2.4A @5V](https://www.lcsc.com/datasheet/lcsc_datasheet_2307171444_etasolution-ETA9741E8A_C7465513.pdf)
LCSC costs
1+ $0.2077
50+ $0.165

amcewen, 2024-09-25:
The key features we'd like (though the USB access to the ESP32 isn't a dealbreaker) are:

### USB access to both the charging chip (to negotiate higher currents) and the ESP32
That'd be nice, but 99% of the time the board will be on a bike and not plugged into a computer; and once WLED is on it then it'll be upgradeable over WiFi.

* The TI chip has support for multiplexing the USB lines so we can do USB to both the charger and to the ESP32.
* The IP5310 wouldn't support this
* The ETA chips might support this, depending on whether we needed to add resistors for higher current charging; as that might interfere with the ESP32

### USB-C access to support negotiation of higher current (we think/presume)
* The TI chip offers this
* The IP5310 offers this (and can negotiate output power at higher voltages too?)

### Direct USB 5V to output 5V alongside charging
So we can run the system while plugged into a power bank for extra running time.

* We *think* the TI chip can charge and run power to the output straight from the charging 5V simultaneously.
* It looks like the IP5310 would maybe run power straight from the charging 5V to the output 5V - depends if the mosfet control would switch off in those cases.  If it doesn't, we can use the resistor "trick" for USB2 to negotiate ~2A charging (which would mean that we **don't** actually need USB-C support) and the MOSFET and diode bypass
* For the ETA chips, we'd need to ~~add the bypass 5V with a MOSFET and diode, as we did in the second-version design~~ see if it can cope with a drain on the supply at the same time as it's charging

### Other notes
* The IP5322 doesn't seem to offer anything better than the IP5310 and is more expensive.
* The TI is more expensive and needs a multiplexer chip, so that seems to have itself out of the market
* The ETA chips are cheapest, ~~would need a MOSFET, diode and resistor to provide the "direct 5V" option~~ and use the same pin for 5V in and out, so maybe has some switching internally to decide whether or not to run the boost circuit?
* The IP5310 would need two MOSFETs, a transistor and a zener diode; and is ~$0.30 more than the ETA.  So with the added components that'd be another £2 or so on the RRP.
* There seem to be some boards on Seeed/Aliexpress using the ETA9740, but I couldn't find anything using the ETA9741

amcewen, 2024-09-25:
I've ordered [one of these boards](https://www.aliexpress.com/item/1005007291941402.html) so we can try out the ETA9740

huffeec, 2024-09-25:
As far as I can tell [this](https://docs.rs-online.com/5199/A700000011325029.pdf) is the data sheet for that board

amcewen, 2024-09-25:
And I've ordered [an IP5310 breakout too](https://www.aliexpress.com/item/1005007457766086.html)

amcewen, 2024-10-07:
It's using an IP5328 but here's a talk about someone building a circuit using it to build a crazy power bank: https://berlin-ak.ftp.media.ccc.de/events/mrmcd/mrmcd24/h264-hd/mrmcd24-330-eng-Designing_a_power_bank_inside_a_can_of_Volt_Cola_hd.mp4

huffeec, 2024-10-11:
Just as a reference [Carl Bugeja](https://youtube.com/carlbugeja)'s project [Coin cell](https://microbots.io/products/codecell) uses the [BQ24232](https://www.ti.com/product/BQ24232) I'm not sure how useful it is for us (Max output current 1.7A)

huffeec, 2024-10-11:
Initial tests of the ETA on the Seeed board

It can charge and provide output at the same time, but.

Disconnecting the usb input causes a brief output interruption ~1 second
Connecting power input while an output load is present prevents charging. So the load needs to be turned off if an input is applied

The IP5328 appears to offer what we need.

It allows input and output simultaneously and there is no interruption if the input is removed.

Running a test to make sure it detects when a battery is fully charged.

amcewen, 2024-10-11:
FYI this is the [datasheet for the usual batteries we'd pair with it](https://cdn.shopify.com/s/files/1/0174/1800/files/PKcell_ICR18650_6600mAh_Final.pdf?11497089046215477949)

huffeec, 2024-10-11:
Charging light still flashing after several hours. Will test again next week.

Battery voltage has reached 4.14V

amcewen, 2024-10-14:
Stuck it on charge again this afternoon and checked in on it about an hour, hour-and-a-half later and the lights were all lit and had stopped flashing.  Battery voltage was at 4.138V on my multimeter at that point, so presumably it was pretty close last time.

amcewen, 2025-01-03:
The prototypes with the IP5310 have been working well (although we probably want to add a power LED or similar to stop it auto-powering off, but that's covered in #31)

closed 2025-01-03 by amcewen