mcqn/my-bikes-got-led #28: My bollards got LED extension

huffeec, 2024-10-30:
From our visit to mini roads a suggested game was light up wands that kids visit in order. This lead to a discussion about if the My Bike's board would be suitable for powering lights and using the ESP C3 as the communication to the microbits. They could communicate with the Microbits over Bluetooth (as opposed to 'Radio')

It would be possible to reduce the width of the new revision to <50mm so that it would fit inside Pex tubing. 

For this we would need:

A second USB footprint (Unpopulated) 90deg around the board for use in the bollards to recharge the battery.

Move/remove the slot to accommodate the optional usb socket placement. Possibly by also moving the mounting holes.

Of course this may be madness but the issue is open to record it just in case..

amcewen, 2025-01-03:
We should get some prototypes with the USB in the "wand orientation" populated for the next run, partly to confirm this approach works (given the "normal orientation" with unpopulated footprints has worked fine on the prototypes), and partly because we need to make some wands :grinning:

amcewen, 2025-01-31:
We got some prototypes made in the "wand orientation" and they work nicely!

huffeec, 2025-01-31:
Needed to complete prototype bollard
Hardware
- [ ] Buy plumbers and caps 2x per bollard
- [ ] Source pipe
- [ ] 3d print (?) hook to hold the end of a single strip of neopixel
- [ ] 3d printed insert to mount board and battery 

Software IDF
- [ ] Complete basic comms between My Bikes boards and Microbits. Initial tests with IDF  and Zephyr respectively has worked. @huffeec  to tidy and create folder in this repo
- [ ] Establish a measure of RSSI as a proxy for proximity 
- [ ] Implement LED driving
- [ ] Receive and respond to Set/Reset/Check game events.
- [ ] Create a mesh network of bollards **?
** Test range of comms without a network first

Software Zephyr (In [https://github.com/mcqn/bike-head](https://github.com/mcqn/bike-head) )

- [ ] BLE advertising for proximity testsing
- [ ] Set/Reset/check modes for game  A, B A+B button events

amcewen, 2025-02-05:
The end-caps look decent.  The key dimensions from them are: OD 55.6mm, ID 50mm (although the aperture for the cap end has an ID of 45.6mm).  The coupler has an OD of 61.5mm.

So we need to find some PEX or similarly translucent pipe for the rest of it.

huffeec, 2025-02-07:
From the site visit this morning. It looks like the perimeter path will be too larger for the range of the bollards, but the two tarmac areas either side of the school are available to us.

[image]

huffeec, 2025-02-07:
This [Bluetooth developer study guide](https://www.bluetooth.com/bluetooth-resources/bluetooth-le-developer-starter-kit/) has a worked example of Microbit proximity service advertising. Although building for the current version fails with Kconfig errors.

For the MBiGL side the [ESP idf proximity sensor](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/ble_proximity_sensor/proximity_sensor_cent) listens for the same support for the Link Loss service (0x1803) as primary service UUID

This builds and runs on our hardware, but currently fails because the service I've set up (a  basic [beacon](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/beacon) in Zephyr is advertising the correct service but not providing the required response.

amcewen, 2025-02-10:
Looking at options for a light-diffusing tube...
* [Clear PVC, OD= 50mm av.bore =42.6mm](https://www.plasticpipeshop.co.uk/50mm-Clear-PVC-Pipe-25m-Length_p_1623.html)
* [Clear PVC, OD= 63mm av.bore =57mm](https://www.plasticpipeshop.co.uk/63mm-Clear-PVC-Pipe-25m-Length_p_1626.html) (so this should fit over the end-cap, rather than into it)
* [Clear acrylic, available in 50mm or 60mm OD, with 2, 3,4 or 5mm wall thickness](https://www.simplyplastics.com/catalog/acrylic-tube/clear-extruded-acrylic-tube/c-24/c-81/p-216).  So the 60mm with 2mm wall would fit over the end-cap; the 50mm inside it.
* [Frosted acrylic, available in 50mm with 3mm wall thickness](https://www.simplyplastics.com/catalog/acrylic-tube/frosted-extruded-acrylic-tube/c-24/c-81/p-302) (available in 60mm too, but the ID would be too small to fit over the end-cap)
* [Clear polycarbonate tube](https://www.theplasticshop.co.uk/polycarbonate-tube.html) available in 50mm or 60mm OD with a variety of wall thicknesses.
* [100% recycled clear acrylic tube](https://www.plastock.co.uk/products/eco-acrylic-cast-clear-tube?variant=42888277786848), available in 50mm OD (the minimum wall thickness is 3mm, so the 60mm OD would have too narrow an ID)
* [Opal frosted polycarb tube](https://www.plastock.co.uk/products/polycarbonate-frosted-opal-tube?variant=43481549340896), 50mm OD with a 2mm wall (they seem out of stock of their 60mm OD with 2mm wall)

I think the frosted ones would be the best option for diffusion, and polycarb is going to be more resilient than acrylic

amcewen, 2025-02-10:
I've ordered 4m of the frosted polycarb.

huffeec, 2025-02-12:
Battling with the [Proximity sensor](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/ble_proximity_sensor/proximity_sensor_prph/README.md) requires an active Link loss service broadcasting. The example in the tutorial above won't compile with newer Zephyr and was proving tricky. 

@amcewen has successfully installed [ble_adv_scan_combined](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/hci/ble_adv_scan_combined ) on our boards and detected a Microbit running [Peripheral](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/peripheral)

I'll add copies of these two samples to the repo as a starting point for implementing .

### Microbit

#### Controller x1

Manages the game running with button presses

- A `Set `the game: Broadcast to all bollards to enter waiting mode
- B `Check`: Broadcast to all the bollards to check status and display correct/incorrect indicator
- A+B `Reset `Broadcast to all the Parcels to generate a random destination number 1-4 and broadcast to bollards to enter waiting mode

#### Parcel x6
Broadcast destination number to bollards
Listen for reset command from Controller 

### MBiGL (ESP32) 

#### Bollard x4

Have a fixed bollard number 1-4

Separate tasks to manage Bluetooth comms and display

**Bluetooth**
Maintain list of Parcel devices within given range (RSSI) over the last five seconds and if they are at the correct bollard or not

Listen for Set/Check/Reset broadcasts from controller. Use a hop counter when rebroadcasting this signal to other bollards to extend the range

On `Set `call display to enter waiting mode
On `Check` call correct/incorrect display based on state of Parcel devices seen list
On `Reset` enter waiting mode

**Display**
Receive calls to set lights based on game state. Ideally animated sequences

huffeec, 2025-02-14:
[f405a26](https://github.com/mcqn/my-bikes-got-led/commit/f405a26a3102af75f5cde4f48f63300b642a1339)
 ·has a working BLE service advert with a 'data' payload which can be set to the game states check/set/rest. Will probably also need to be an additional state for 'in play'

Basic microbit button code is working, but my initial attempt at polling an A+B press isn't

huffeec, 2025-02-17:
ESP example code separates the name (0x09) and the date (0x24) in the large array that is produced by a scan event,

In the short term we can just scan on 0x24 and record the data and mac address to determine controller and parcel devices and their state. 

[image]

amcewen, 2025-02-19:
I've added a new task for the LED display, which runs a rainbow strobe at present.  Also added `main/app_led.h` which has the API that the rest of the game should use.  Right now it just prints out to the log that it was called, and returns an `-ENOTSUP` error code.

I don't think the calling code (in this case) will care about any return codes from it, but it was more for futurproofing.  The calls are:
* `LEDs_Start()` - this starts the task and sets up the Arduino subsystem.  It should be called somewhere near the start of `app_main` (before it gets into a loop at least)
* `LEDs_Success()` - displays a rainbow to show the success condition
* `LEDs_Failure()` - displays a red, slow flash(?) to show a failure condition
* `LEDs_Waiting()` - displays a yellow "game in progress" condition.  Maybe a larson scanner, maybe just fading up and down in brightness?
* `LEDs_Progress(uint8_t aPercentDone)` - for debugging more so than the game itself, fills the strip with colour to the given percentage.

amcewen, 2025-02-19:
I've added a basic implementation of the display now.  All of the API functions listed above work, and the strip animates for each state.  It starts up with the display off.

* Failure has the whole strip fade up and down in brightness of all-red
* Success is the FastLED cylon rainbow
* Waiting is a yellow Larson scanner
* Progress lights the given percentage of LEDs (and fades out higher ones if the level drops, rather than going straight to off; seems a bit smoother in operation)

I've got a bit of simple code that sets it to failure if the signal strength is too low; then progress above that; and success if it gets strong enough.  However, that's in `app_bt.c` and I couldn't remember if you still had changes in that to commit @huffeec, so I've left that not-commited for now.  It's pretty trivial:
```
if (rssi[i] < -90)
{
    LEDs_Failure();
}
else if (rssi[i] < -50)
{
    LEDs_Progress(100+rssi[i]);
}
else
{
    LEDs_Success();
}
```

I've also added a call to `LEDs_Waiting();` after the call to `LEDs_Start();` to (a) test that state, and (b) show something at startup for now.

huffeec, 2025-02-21:
Current state of parcel and controller boards.

Parcel: On start/reset the board picks and broadcasts an 0x24 response message with 'BOL' and a random number between 0-3
The Micro:bit reset button acts as the game reset for this parcel and selects a new BOL num. 

Controller:  On start the boards broadcasts an 0x24 response message with  'Check' as the data payload.
The Micro:bit buttons toggle the state of the game between 'Set' (A) or 'Check' (B) by changing the response message payload to 'Set'/'Check'

Both boards will need laser cut shapes to hold them and the battery. The controller will need the functions of the buttons etched onto it. The parcel board just needs to hold the battery and board. A hole to give access to the reset button is needed.