mcqn/my-bikes-got-led #30: r20241810 board bringup

amcewen, 2024-11-21:
We've had five of the `r20241810` boards fabbed.  This issue is for notes and anything that needs addressing before the next production batch:

* [ ] The on/off switch is inverted
* [ ] The terminal block connector needs to be 1.1mm nearer to the edge of the PCB, as the plug-in block fouls a tiny bit against it
* [ ] Double-check the LED part number for the charger LEDs - the JLCPCB ones are side-emitting (which work but the spacing ends up a bit weird-looking with the charlieplexing)
* [ ] Check the WiFi LED part

amcewen, 2024-11-21:
* We haven't got the USB working yet, so need to program the board via the "FTDI" serial header.
* The LED data line is on `GPIO 5` so needs to be configured in WLED to bring the lights to life

amcewen, 2024-12-19:
Got the scope set up to monitor `D+` and `D-`.

This is a Museum in a Box board as a reference:

[image]

The straight-from-production My Bike's Got LED board seems to have half the amplitude that we'd want on the `D+` line:

[image]

And it seems not very square!

Tapping into the USB (rather than ESP32-C3) side of the resistors with some wires to a USB-C breakout board gives a much squarer trace with the right sort of amplitude on both `D+` and `D-`:

[image]

However, it fails in seemingly the same manner on the Linux side.

We tried programming the ESP (over serial) with the CodeCell getting started example, just to ensure nothing was using the USB or other important pins, but that made no difference (beyond us spotting issue #31)

We set the trigger on the scope to catch when the USB protocol activity starts (when `D+` went high for the 2nd or 3rd time, depending on the exact trigger level for it) and then hand-decoded the USB traffic, referring to [this USB explainer for voltage levels and bits](https://www.engineersgarage.com/signal-and-encoding-of-usb-system-part-5-6/) and [this one for how those bits map into USB packets](https://beyondlogic.org/usbnutshell/usb3.shtml)

There seem to be a bunch of `ping` packets, to different addresses, but after an initial intact packet we then saw one with the last bit of the `PID` flipped (and so invalid) followed by one that looks like a retransmission with an intact `PID`.  The invalid-then-intact pairings were each for different addresses, and all were at 1ms intervals (as we'd expect to stop the device going into suspend).  We didn't capture enough data in the scope memory to get beyond the pings (we saw about 8 of those IIRC)

Would be good to repeat the exercise monitoring a Museum in a Box board to see what the successful bring up of USB looks like.

amcewen, 2024-12-19:
I was thinking some more overnight about how the "invalid ping" packets were always wrong by the last bit and how it seemed strange for it to be so regular.

I think our decoding was off by 1-bit.  If what we thought was the last bit of the sync byte was actually the first bit of the PID then we'd have PIDs that were always 10100101.  So they're all NAKs rather than pings.

huffeec, 2024-12-20:
Equally  out by two would give a consistent 1101 which has the correct form for the SOF token packet.

amcewen, 2024-12-24:
Had a look at the Museum in a Box USB traffic today.

Got lots of SOF packets at the start, at 1ms intervals.
* `00000001 10100101 01110011 01111110`
* `00000001 10100101 11110011 01100000`
* `00000001 10100101 00001011 01110111`

First capture wasn't getting anything beyond those.  I think that's because it's still only ~8ms after insertion of the cable, so maybe the ESP32 is still starting up.

The traces on from the Museum in a Box board look like this, as another reference.

[image]

Switching to triggering on a *falling* `D+` (well, the one that's high before the packet starts) and then the Nth edge seems roughly to be the Nth packet.  I think the rest of the traffic is too quick for the triggering to catch properly (or at least I can't work out the right parameters to catch it) on the rising edge.

Skipping 25 on the falling edge seems to catch something interesting:

* some 1ms interval SOF packets...
* `00000001 10100101 00001110 00100001` another SOF packet
* 1ms delay then at -5.6us - `00000001 10110100 00000000 00001000` - Setup token, I think?  Address 0, endpoint 0
* at -2.32us - `00000001 11000011 00000001 01100000 00000000 10000000 00000000 00000000 00000010 00000000 10111011 00101001` - a DATA0 packet, 8 bytes(?)
* at 6.32uS - `00000001 01001011` - an ACK
* at 8.84us - `00000001 10010110 00000000 00001000` - an IN token (host wants to read from the device).  Address 0, endpoint 0
* at 12.16us - `00000001 11010010 01001000 10000000 00000000 01000000 11110111 01000000 10000000 00000010 01011100 00001100 10000000 00001000 10000000 10000000 10000000 01000000 11000000 10000000 00000010 01101010` - a DATA1 packet, 18 bytes(?)
* at 27.46us - `00000001 01001011` - an ACK
* at 29.39us - `00000001 10000111 00000000 00001000` - an OUT token (host wants to send to the device).  Address 0, endpoint 0
* at 32.66us - `00000001 11010010 00000000 00000000` - a DATA1 packet, 0 bytes I think.
* at 35.97us - `00000001 01001011` - an ACK
* at 257.3us goes into SE0 (both D+ and D- low) until at least 21.41mS - capture ended then.

Trying the modded MBiGL board, capturing after 12 edges, otherwise it seemed to just get the SE0 trace for all of it (if I tried any higher)

It was actually plugged into the onboard USB-C socket though!  The probes were onto the wires plugged into the breadboard with the USB-C breakout; and the battery was charging (just the final LED flashing IIRC), in case that affects things?

[image]

* Some 1ms interval SOF packets...
* at 50ms - `00000001 10100101 10101110 10010011` - an SOF packet
* at 50.66ms - `00000001 10110100 00000000 00001000` - Setup token?  Address 0, endpoint 0
* at 50.67ms - `00000001 11000011 00000001 01100000 00000000 10000000 00000000 0000...00010 00000000 10111011 00101001` - a DATA0 packet (the `...` were some more 0s that I didn't work though decoding the exact count)
* at 50.68ms - `00000001 10110100 00000000 00010000` - another Setup token??  That's address 0, endpoint 1 - but that might be a mis-decoding?
* at 50.69ms - `00000001 11000011 00000001 01100000 00000000 10000000 00000000 00000000 00000010 00000000 10111011 00101001` - a DATA0 packet, 8 bytes.  Identical to the last DATA0 packet, and to the one from the Museum in a Box capture
* at 50.698ms (a noticeable gap compared to the gap between packets so far, or during all of the Museum in a Box capture) - `00000001 10110100 00000000 00001000` - another Setup token?  Address 0, endpoint 0
* at 50.701ms - `00000001 11000011 00000001 01100000 00000000 10000000 00000000 00000000 00000010 00000000 10111011 00101001` - another identical DATA0 packet
* at 50.99ms - `00000001 10100101 01101110 10010001` - back to SOF

There's then another burst of traffic at 51.01ms - I need to pick up decoding again there.  The capture is saved on the scope as `mbgl0.bin`

However, from a skim over it, there are two or three more bursts of traffic and then it drops to SE0 state.

From some further, higher-level decoding, it seems that a "Setup stage" is a Setup packet followed by a DATA0 packet, both sent by the host, and then an ACK (if it was received okay) from the device.

In the Museum in a Box example, that ACK comes through, so things move onto a "Data stage": that's an IN packet from the host to ask for some data, a data packet from the device, and then an ACK from the host; followed by the "Status stage", where an OUT packet is sent from the host followed by a zero-length data packet from the host, then an ACK from the device.

The My Bike's Got LED trace seems to show no response from the ESP32, and the host retrying the "Setup stage" a few times.

[https://www.beyondlogic.org/usbnutshell/usb6.shtml] details the format of the setup packet, which is 8 bytes long.  Decoding it:
* `00000001` - request type: device to host; standard; recipient is device
* `01100000` - request: "get descriptor"
* `00000000 10000000` - a `wValue` of 256, which defines the descriptor type desired
* `00000000 00000000` - index or offset of 0
* `00000010 00000000` - if there's a data phase, 64 bytes should be transferred

amcewen, 2024-12-27:
* tried the board I've been using to see what that's like.  The scope traces look similar: pretty square with a bit of ringing after each transition.  Didn't decode in detail, but could see a similar shorter packet followed by a long, presumably`DATA0`, packet and then a pause before it tries again.  With the pause being the gap where the ESP32 should provide the `ACK`.
* Tried the probes on the ESP32C3 side of the resistors and that looked *very* wrong:
  [image]
  Both `D+` and `D-` looked similar, although on this capture there's only `D+` as I'd removed the probe on `D-` in case it made any difference
* Checked the resistors `R15` and `R16` in the Kicad files.  They're specced as 22 ohm 0402.  The Farnell part numbers assigned to them are the right value but the wrong size.  More importantly for these test boards, the LCSC part number is the correct size, but 2.2k ohm.  That might be the problem :zany_face: 
* Moved the tacked-on wires on the main test board to the ESP32C3 side of those resistors, and the board enumerated on USB first time when I plugged it into my laptop and showed up as `/dev/ttyACM0`.

amcewen, 2024-12-30:
I've ordered some of [these 22R 0402 resistors from Farnell](https://uk.farnell.com/neohm-te-connectivity/crg0402j22r/res-22r-5-0-063w-0402-thick-film/dp/1174143) so we can swap some out and confirm that the resistors are the problem.

amcewen, 2024-12-31:
The resistors arrived today; I replaced the ones on the prototype board I've been testing, and it shows up as `/dev/ttyACM0`, I've seen serial output from it, and I can reprogram it over USB! :tada: 

[image]

huffeec, 2025-01-08:
Replicated this on another board

closed 2025-01-08 by huffeec