mcqn/my-bikes-got-led #31: ESP32-C3 on its own doesn't draw enough power to keep the charger chip on

amcewen, 2024-12-18:
The `r20241810` boards auto power off after ~30 seconds when running on battery if they're just running basic code on the ESP32-C3.

From the datasheet for the charging chip, that's because it's drawing less than 45mA and that triggers an auto shutoff feature.

Maybe we should add a "Power" LED to bump up the minimum current draw sufficiently to keep it running?

huffeec, 2025-01-10:
After struggling to source an led capable of handling this current D25 has been removed. R27 set as R120

There is a [register setting](https://github.com/Andyzug/PiBat_IP5310/blob/main/IP5310%20REG%20V1.2.zh-CN.en.pdf) that extends the 'boost low current shutdown time' but the maximum is 63mins.

SYS_CTL2 0X0E 2:0

amcewen, 2025-07-18:
Checked it again with the latest production boards and it seems to continue running well.

closed 2025-07-18 by amcewen