mcqn/my-bikes-got-led #34: Apply for OSHW certification

huffeec, 2025-01-20:
Following the process from [My Baby's got LED](https://github.com/mcqn/my-babys-got-led) applying at [https://application.oshwa.org/apply](https://application.oshwa.org/apply)
We will also need photographs of the new boards

- [ ] @huffeec Draft
- [ ] @amcewen updated/approved
- [ ] @huffeec Applied

**BASIC INFORMATION
SECTION 1 OF 4**
This part of the form asks for basic information about the responsible party for the project to be certified, such as the name of the individual, company, or organization certifying the project and contact information for any future correspondence from OSHWA regarding your certification.

This certification is on behalf of a: Company
Name of Individual, Company, or Organization Responsible for the Certified Item: MCQN Ltd
If not an Individual, name of Individual with Authority to Bind the Company or Organization: Adrian McEwen

**Contact Information**
Country: United Kingdom
Street Address Line 1: DoES Liverpool, The Tapestry
Street Address Line 2: 68-76 Kempston Street
City/town/village: Liverpool
State/Province/Region: Merseyside
Zip code/postal code: L3 8HL
OSHWA Contact Email Address: redacted
Public contact email address: [info@mcqn.com](mailto:info@mcqn.com)

**PROJECT INFORMATION
SECTION 2 OF 4**
This part of the form asks for information about the project you wish to certify. Information submitted here will appear on your project’s profile listing.

Project name: My Bike's Got LED
Project version: r20250110
Have you previously registered a version(s) of your project with OSHWA? No
Project description: My Bike's Got LED is a portable LED driver board using WLED to drive individually addressable LEDs at 5V from LIPO batteries. It incorporates a battery charging circuit over USB which also provides an input for a secondary battery. It uses an ESP32 C3 mini chip and can be controlled using its own WiFi network or by connecting to an existing network. The development and documentation for the board is available on github, and the board is manufactured by mcqn and sold on Tindie.
Project website: mcqn.com/ibal234
Primary Project Type: IOT
Additional Project Types: Arts, Electronics, Home Connection
Project Keywords: WLED, MQTT

Where can the documentation be found for your project?: https://github.com/mcqn/my-bikes-got-led

- [ ] All project documentation and design files are available in the preferred format for making changes.

Does your project incorporate or build upon other open projects that are not currently certified by OSHWA? If so, use this space to cite those projects

|Citation Title | Citation URL|
|-- | --|
|WLED | https://github.com/Aircoookie/WLED|

**LICENSING INFORMATION
SECTION 3 OF 4**

This part of the form asks for information about your project’s 
licensing. In order to qualify for OSHWA certification, you must have 
chosen an open source license for your hardware, your software (if any),
 and your documentation. Your licenses for each of these elements will 
appear on your project’s profile listing.
What license is your project's hardware licensed under? CERN
What license is your project's software licensed under? MIT
What license is your project's documentation licensed under? CC BY-SA

|Statement | Response | Explanation|
-- | -- | --
|The project is licensed in a way to allow for modifications and derivative works without commercial restriction. | Yes |  
|There is no restriction within my control to selling or giving away the project documentation. | Yes |  
|Where possible, I have chosen to use components in my hardware that are openly licensed. | Yes |  
|I understand and comply with the "Creator Contribution requirement," explained in the Requirements for Certification. | Yes |  
|There is no restriction on the use by persons or groups, or by the field of endeavor. | Yes |  
|The rights granted by any license on the project applies to all whom the work is redistributed to. | Yes |  
|The rights granted under any license on the project do not depend on the licensed work being part of a specific product. | Yes |  
|The rights granted under any license on the project do not restrict  other hardware or software, for example by requiring that all other  hardware or software sold with the item be open source. | No | The WLED software uses a MIT licence and can be used in other  non-open projects, but the hardware is restricted under the CERN Open  Hardware Licence Version 2 - Strongly Reciprocal.|
|The rights granted under any license on the project are technology neutral. | Yes |  

**CERTIFICATION
SECTION 4 OF 4**
This part of the form asks you to certify that you agree to specific terms of the [OSHWA Certification Mark License Agreement](https://certification.oshwa.org/license-agreement).

You agree to the following terms:

I have provided OSHWA with accurate contact information, recognize that all official communications from OSHWA will be directed to that contact information, and will update that contact information as necessary.
I will only use the certification mark in compliance with official certification guidelines.
I acknowledge that all right, title, and interest in the certification mark remains with OSHWA.
I acknowledge that OSHWA has the right to enforce violations of the use of the mark. This enforcement may involve financial penalties for misuse in bad faith.
I have the ability to bind those responsible for the certified item to this agreement.

If you do not agree with any of the above terms, please explain.
Provide a brief explanation (500 characters)
Relationship to certified item: I am the primary developer of the certified item, as well as a manufacturer and retailer of the item as a commercial product.

I agree to the terms of the OSHWA Open Source Hardware Certification Mark License Agreement, including the Requirements for Certification and Usage Guidelines incorporated by reference and including license terms that are not present in or conflict with this web form. I acknowledge that by agreeing to the terms of the OSHWA Open Source Hardware Certification Mark License Agreement that I am binding the entity listed to the License Agreement. I recognize that I will receive my unique identification number that allows me to promote my project as OSHWA Open Source Hardware Certified in compliance with the user guidelines via the email provided to OSHWA after submitting this form.

If you are the parent or legal guardian entering into this agreement on behalf of an individual under the age of 18, please provide your name to certify that you also agree to be bound by this agreement.: Not applicable.

amcewen, 2025-02-16:
Looks good.  I've submitted it, given I was already looking through it.  As above, though I added "Wearables" and "IOT" to the "Additional Project Types" section.

huffeec, 2025-04-07:
Certified as UK000073
https://certification.oshwa.org/uk000073.html

closed 2025-04-07 by huffeec