mcqn/my-bikes-got-led #35: My bollards got led - 3d printing

huffeec, 2025-02-12:
Breaking out from #28 for the mechanicals

Pipe and connectors have been bought, to complete the build we will need:

- [ ] 50mm opaque pipe to make a sleeve at the bottom of the bollard to contain the electricals.
- [ ] Plug piece to fit into the base and hold the bollard. This piece needs to allow access to the bottom connect so that it can be unscrewed for the power switch and charging.
- [ ] Holder for battery and board to fit within the tube. Needs to have access to the usb port for charging and accommodation for a toggle switch
- [ ] Hook for top of tube to hold lights in place.

amcewen, 2025-02-15:
Made a first pass at the plug piece.  It's on the printer at the moment, and will be done in about 9 hours.

amcewen, 2025-02-17:
Screwfix won't sell the 50mm pipe from the Islington branch, but when discussing it we decided actually just to print all of that section.

Also made an assortment of tweaks to the sizes after testing the first print:
* made the "finger gap" round the cap wider for better access
* slimmed down the outer diameter slightly, as it wouldn't quite fit into the base
* narrowed the inner hole so the pipe shouldn't wobble when fitted

It's on the printer now, going to be a 21 hour print in total!

amcewen, 2025-02-19:
The new plug/sleeve works really nicely.  The second print failed as it ran out of filament (turns out the 32m of filament it needed is more than "a lot" when you're looking at what's left on the spool.  There are two new prints running with fresh 1kg spools on the DoES Liverpool Ultimakers at the moment.

I've cut the pipe for the first bollard - 66cm long, and the two LED strips to go inside - each has 30 LEDs (so is 50cm long); the LED strips are wired into the `data` and `clk` pins on the My Bike's Got LED board.

With the LED strips removed from the waterproofing sleeve, they short if they're placed back-to-back and it's tricky to pull them taught inside the pipe.  I've cut a strip of polyprop (checked in as `light-backing.svg`) to act as a backing; I think they'll stick pretty well to that with double-sided tape (only tried a couple of short sections to tack it in place initially).  For future builds we should get the adhesive-backed LED strips to begin with 😁 

The polyprop has a small hole in each end, which I think will make it easier to fasten in place.  It seems to work okay with the hook 3D print and a Post Office elastic band, although that's a little long.

Still needed for the first prototype:
* [ ] Soldering switch to the My Bike's Got LED board
* [ ] Sticking the LED strips to the carrier polyprop
* [ ] Soldering LED strips to the My Bike's Got LED board
* [ ] Working out how to secure the bottom end of the LED strip/carrier
* [ ] Glue the pipe parts together (I got a tub of ABS solvent weld from Screwfix, feels like the first thing to try?)
* [ ] Flashing code to it

huffeec, 2025-02-21:
From the first test outside today it's become clear we need the light strips attached to the side of the tubes. They don't give off enough light to be clearly visible if it is defused.

For this both [lastmile_boardHolder.FCStd](https://github.com/mcqn/my-bikes-got-led/blob/main/my-bollards-got-led/physical/lastmile_boardHolder.FCStd) and [lastmile_tophook.FCStd](https://github.com/mcqn/my-bikes-got-led/blob/main/my-bollards-got-led/physical/lastmile_tophook.FCStd) need reworking. The polyprop strips will still need to be under some tension to keep the strips against the tube walls all the way down.

amcewen, 2025-02-25:
I've extended the pegs on the LED tensioner disc (went as high as 5mm but that stuck out too much to go into the tube) to 4mm, and flared the extrude by 10&deg; which means (at 5mm extrude at least) that the LED backing strips stay on them once attached.  And added the notch for the lower version to fit into the spacer piece.

I also trimmed the board-to-battery disc by 1mm, as that was too much of an interference fit with the tube.

amcewen, 2025-02-25:
Did some trial fitting, and it's coming together nicely.  A few tweaks:
* The overall length is a bit long, if we can lose 9-10mm from the battery "compartment" that would help
* We should shorten the LED backing too
* Some holes in the piece between the battery and the PCB would be useful, to allow wires to pass through

The battery test ran on the rainbow pattern (with three strips) for 9h30m before I turned it off.

huffeec, 2025-02-26:
Remaining printing

- [ ] [lastmile_boardBrace.FCStd](https://github.com/mcqn/my-bikes-got-led/blob/main/my-bollards-got-led/physical/lastmile_boardBrace.FCStd) x2
- [ ] [lastmile_boardBottomPlug.FCStd](https://github.com/mcqn/my-bikes-got-led/blob/main/my-bollards-got-led/physical/lastmile_boardBottomPlug.FCStd) x3
- [ ] [LED-tensioner-top.FCStd](https://github.com/mcqn/my-bikes-got-led/blob/main/my-bollards-got-led/physical/LED-tensioner-top.FCStd) x2

amcewen, 2025-03-12:
The LED tensioner disc has a tendency to skew when it's fed into the tube, and the bottom one is totally inaccessible to try to straighten out.

I think if we made it thicker it would then seat properly in the tube and not have any room to twist.

amcewen, 2025-03-13:
I made some notes on how to assemble things in a `README.md` in the folder with the 3D design files.  Needs the names updating, and links adding, to the various 3D files (and tweaking if we find more things to note when assembling the remaining 3 bollards)