mcqn/my-bikes-got-led #37: Make container for sound reactive components of My Bike's got LED

huffeec, 2025-03-05:
Container to include

- My bikes got led board
- PCM1808 ADC [breakout](https://www.aliexpress.com/item/1005005819147396.html)
- Panel mount 3.5mm jack socket
- On/off switch
- Battery
- Nice to have > input jack for handlebar mounted switch to cycle 

Box needs to be robust and weather resistant. 

Possible to remove battery with a bracket underneath the main case suitable for a four cell pack. It should be easy to remove the battery without pulling on the wires. This may require MBiGL boards with the vertical mount JST's

I've got in touch with Peloton to arrange having a look at the current setup.

amcewen, 2025-03-05:
I've ordered another of the PCM1808 breakouts, and some more panel-mount jack sockets so we can make up a second container.

The weather-resistant side of things *could* be it getting put into a saddle-bag again (doesn't have to be, just noting it as still an option)

closed 2025-04-07 by huffeec

huffeec, 2025-09-26:
The pink box is in for repair.

The wires to the switch had become detached and removed copper traces near sw3 & q1. This also removed p1 of q1. I have replaced q1 and resoldered the switch wires.

It is possible to connect to wled on the DoES network. The sound reactive option was unticked so I can't see which order the audio pins need connecting in. Will check with the white box before I reconnect the new ADC.

huffeec, 2025-10-17:
I've replaced the board in the pink box because the fix described above didn't hold. Because of this there is an extra pin out on the header (P10) so I've used this for the button instead of desoldering the wifi led.