mcqn/my-bikes-got-led #38: Shuffle pins for the I2S input to match footprint for INMP441 microphone?

amcewen, 2025-03-05:
It would be nice if there was an even simpler way to add an I2S mic, and given the popularity of the 2 rows of 3 pin 0.1" headers on the small circular INMP441 microphone, it'd be nice if the relevant pins on the MBiGL PCB were lined up so you could solder in two rows of headers and the the microphone would just plug in.

huffeec, 2025-03-25:
https://github.com/mcqn/my-bikes-got-led/commit/d60d5c8d278dbaae7465e52ee73bb3c98afd6561 has this arrangement for the header. Needs a more intuitive silkscreen and sense check the pins used for the microphone and remote button.

huffeec, 2025-05-02:
Silk screen tidied and pins added to [readme](https://github.com/mcqn/my-bikes-got-led/tree/main/electronics)

amcewen, 2025-10-05:
Ordered a batch of boards with this on, so assuming it works, we'll then be able to close this.